TheZeroBeast/TZB-MMU2S-Firmware #239: Cam unlocking

pennystockss, 2022-09-23:
Ensure you're running the latest **RELEASE** and you've checked the **Wiki** for answers before reporting. Any issues reported regarding previous **RELEASES** will automatically be closed and not responded to, this ensures the developer maintains focus & streamlines the development process.

**TZB-MMU2S Release Version**
Current 4.0.2 on stock MK3S+/MMU2S+

**Describe the bug**
While loading or unloading during a tool change, sometimes the idler motor unlocks and the filament rotates the ilder, causing the teeth on the MMU2 to not push the filament. Eventually errors out with un/load error.

I haven't seen it do it directly, but 3 times on the past 2 prints, the MMU2 was in error. The filament was somewhere in the tube going to the extruder at some random point, not stuck. When trying to move filament back or forth with the left or right button, it either does nothing or moves filament in another slot because the idler is out of place and not pushing on the correct filament.

Further inspection shows the idler is loose, not locked in position. I can spin the idler with my fingers. When I push a button, the stepper motor does move the idler to try to do it's function, then gives up and idler unlocks.

I believe the stepper going to the idler unlocks for some reason while it's loading/unloading filament.

**To Reproduce**
Unknown. Out of maybe 500 tool changes it happened 3 times.

**Expected behavior**
idler to be locked when tool changing.

**Screenshots**
n/a

**Serial Log**
Don't have

**Additional context**
First time using TZB firmware. Never had this issue in 1+ year on stock firmware.

pennystockss, 2022-09-24:
I caught it! I would walk in every now and then when I heard it start to change. That is what it does but the idler is either loose and snags on some friction then spins or the stepper for the idler tweaks and spins a bit during loading. The stepper is locked during successful filament changes. I'm not sure if the stepper motor unlocks and grabs on friction or stepper tweaks but every 100 or so filament changes, the idler rotates during loading and releases tension failing to load.

timemaster5, 2022-10-17:
I am having the same issue too. As I don’t print multi colour often, I see it mainly at the end of the print session. After many printing hours I find an mmu unload error with the fillament in the middle of extrude tube.

timemaster5, 2022-10-17:
Could you check with the 4.0.2 version? The cause can be different in the previous release.

floridaservices, 2022-10-17:
Will be a bit but yes it has been on my list to upgrade

pennystockss, 2022-10-17:
I was on 4.0.2. Unlike Floridaservices, mine never recovers. It unload fails with the filament just hanging out in the tube between the selector and extruder. I might actually try 4.0.1 just to see if it recovers. I love the features of TZB FW. I have an 9hr, 240 tool change print I'm about to do, I'll update after I test.

floridaservices, 2022-10-17:
By recovers, I meant I can interact with it and have it try the unload again and it completes the unload and continues the job.  I am sorry for the confusion

pennystockss, 2022-10-17:
> By recovers, I meant I can interact with it and have it try the unload again and it completes the unload and continues the job. I am sorry for the confusion

Oh that's a bummer. I'm trying 3.3 since you said that, to see if it has the bug.

floridaservices, 2022-10-17:
ehh, TZB beats the stock firmware any day, stock firmware would require hitting the reset button, unloading/reloading filiment, maybe losing my print in the process.  I'll take hitting the button and having it try again!

EurohellDesign, 2022-10-18:
Same thing here. We have about 40 Machines running with TZB Firmware and i could see this issue (and some others) many times, thats really annoying. My solution for now is reverting all machines to 3.3.0 because here this error doesn't occur and there is a way to get the mmu out of the situation not communicating to the einsy any more without losing the entire print.