TheZeroBeast/TZB-MMU2S-Firmware #243: What next?

timemaster5, 2023-10-27:
Prusa, after those all years, got back to the MMU and released version 3 together with a new MK3s firmware. I would like to know the community's opinion and also an idea of the Author of this great firmware, what would be our next move.

Will you merge in the latest Prusa FW for the MK3 and MMU2s (and possibly 3)?
Or is it better to upgrade to MMU3 from Prusa and get back to the original firmware?

Does anyone have a direct comparison of how good or bad the new versions are compared to TZB?

People with custom setups supported here will stay, I guess.

Naugrimohtar, 2023-10-28:
Watching

Tony763, 2023-10-28:
I use TZB FW for almost two years on MK3s at my workplace. It saved me lot of hassle as before it, our machine was totally useless as almost every second print failed. 

For MMU3, only time will show, if it is worth it and if we can stick with original FW or if we will need another custom FW.

I ordered both upgrade kits (MK4 and MMU3) so when it is delivered, I will report there with first sights.

gizzburn, 2023-10-28:
I also used this FW from the start and it was the best out there, I had a Bondtech extruder and TZB made me a custom version, that was superb . But when the latest version came out from Prusa I asked TZB if he was going to update his FW with all the upgrades, He said no because Prusa has used some of his ideas and he was not letting them have anymore idea's. Its a shame but its his decision. I have gone back to using the Bondtech fork of the FW, ok but not as slick as the TZB was.

timemaster5, 2023-10-31:
Thanks to all for your thoughts. This firmware saved all of us here at the time, as the original was unusable, and it is a shame no one credits TBZ for it. And it was a long time waiting for MMU to get the attention again. Let's see what changed in the origin. This project will hopefully serve TZB at least to prove their qualities to grab some excellent job positions or other opportunities. So thank you again, and keep doing a great job, it really helped a lot.

> I use TZB FW for almost two years on MK3s at my workplace. It saved me lot of hassle as before it, our machine was totally useless as almost every second print failed.
>
> For MMU3, only time will [...]

Was it worth to upgrade MK3 to 4?

Anyway, experience is welcome. So far, I have seen only photos of lovely MM prints, which kept coming the whole time. And haven't seen any real reviews yet.

Tony763, 2023-10-31:
> Was it worth to upgrade MK3 to 4?

Can't say. Shamefully both upgrade kits were not delivered, yet. (We are from Czech Republic few km away from Prusa Headquater and ordered kits as soon as pre-orders were enabled.)

timemaster5, 2023-10-31:
Well, I meant financially. It seems too expensive to me. To buy a new MK4 kit and sell an MK3, or keep both, may be an option. But I haven't decided to myself.

TheZeroBeast, 2026-03-09:
Hi all, glad the code brought some benefit. It was a large commitment for a few years there and I still use it without issue. I haven't noticed any firmware improvements that can justify rebasing and testing. If anyone gets to testing feel free to submit pull requests, I'll vet and merge if worthy.